Chadster766/McWRT #83: The new 1.0.8 firmware is constantly pulling IP

azurewood, 2015-01-08:
Jan  8 17:51:08 OpenWrt user.notice root: Get WAN IP address...
Jan  8 17:51:09 OpenWrt user.notice root: Get WAN IP address...

A few hours later to be restarted, otherwise the web network does not work

azurewood, 2015-01-08:
reflash this version again, this getting IP address problem does not occur again, waiting to see

azurewood, 2015-01-10:
it's getting this again after a reboot..

Jan 11 05:28:51 OpenWrt user.notice root: Get WAN IP address...
Jan 11 05:28:52 OpenWrt user.notice root: Get WAN IP address...
Jan 11 05:28:53 OpenWrt user.notice root: Get WAN IP address...
Jan 11 05:28:54 OpenWrt user.notice root: Get WAN IP address...
Jan 11 05:28:55 OpenWrt user.notice root: Get WAN IP address...
Jan 11 05:28:56 OpenWrt user.notice root: Get WAN IP address...
Jan 11 05:28:57 OpenWrt user.notice root: Get WAN IP address...
Jan 11 05:28:58 OpenWrt user.notice root: Get WAN IP address...
Jan 11 05:28:59 OpenWrt user.notice root: Get WAN IP address...
Jan 11 05:29:00 OpenWrt user.notice root: Get WAN IP address...
Jan 11 05:29:01 OpenWrt user.notice root: Get WAN IP address...
Jan 11 05:29:02 OpenWrt user.notice root: Get WAN IP address...

azurewood, 2015-01-10:
Marvell Armada XP RD WIFI Board
Firmware Version	OpenWrt Attitude Adjustment 12.09.1 / LuCI Trunk (git-1c9570b)

the firmware version has changed after the reboot. why?

Chadster766, 2015-01-10:
It must have switched to the alternate image because of an issue with the other firmware. 

Do you have a USB to TTL cable? There is specific instructions to load to both image location. 

If not load the Linksys firmware and then reload v1.0.8.

azurewood, 2015-01-10:
I tried the second method. It works. 
Thank you!

closed 2015-08-17 by Chadster766