Chadster766/McWRT #84: Add MIA (Moms internet Access helper) as package for OpenWRT

raversnet, 2015-01-09:
Not sure if this is the place to ask but since i use this firmware and love it.... :)

I would like to request this package compiled for OpenWRT.  Its a parent scheduling module so i can set blocks for my teenager.  Unfortunately i dont have the platform to compile.  Would love if someone could tackle this and add to the repository.  The author gave me the code to work with.  

If not then thats cool too.  Thanks anyways :)

Code:
https://drive.google.com/folderview?id=0BwhvHRyRV5PlYkp2YlJ6Y3M5Ync&usp=sharing

MIA website: 
http://www.avidz.net/mia/

Chadster766, 2015-01-09:
I will look into it.

rip057, 2015-01-10:
Good luck... also I already have gawk sed and etc for cc compiled and
working beautifully

7LK, 2015-01-10:
great idea

closed 2015-08-17 by Chadster766