Chadster766/McWRT #87: unable to use usb storage is /overlay or /

alirz1, 2015-02-09:
Hi,
im trying to setup a usb falsh stick as my main storage for the router so that i can install opkg packages without running out of space. From what i understand it might be better to user /overlay so thats what im trying to do.

The usb stick itself mounts fine but i cannot seem use for rootfs
im running: OpenWrt Attitude Adjustment 12.09.1 / LuCI Trunk (svn-r10537)

root@OpenWrt:/home/alirz# cat /etc/config/fstab
config global 'automount'
      option from_fstab '1'
      option anon_mount '1'
  
config global 'autoswap'
      option from_fstab '1'
      option anon_swap '0'
           
config mount
      option target        '/overlay'
      option device        '/dev/sda1'
      option fstype        'ext3'
      option options       'rw,sync'
      option enabled       '1'

root@OpenWrt:/home/alirz# df -h
Filesystem                Size      Used Available Use% Mounted on
rootfs                   30.8M     28.4M    852.0K  97% /
/dev/root                37.0M     21.1M     15.9M  57% /rom
devtmpfs                123.6M         0    123.6M   0% /dev
tmpfs                   123.7M    368.0K    123.3M   0% /tmp
ubi0:syscfg              30.8M     28.4M    852.0K  97% /tmp/syscfg
ubi0:syscfg              30.8M     28.4M    852.0K  97% /overlay
overlayfs:/overlay       30.8M     28.4M    852.0K  97% /
/dev/sda1                 1.9G     34.9M      1.7G   2% /tmp/overlay-disabled

Chadster766, 2015-02-09:
I think @tusc figured this out :smile: 
https://github.com/Chadster766/McWRT/issues/64

alirz1, 2015-02-10:
Thanks. But I know the pong config method to use it is main storage.
Now I'm trying to mount /overlay on the sub but always end up with /overlay-disabled.
I've tried all co figs to use as rootfs. Doesn't that not work with mcwrt?

If you don't mind please see my post

https://forum.openwrt.org/viewtopic.php?id=55665

Chadster766, 2015-02-10:
Sorry I don't know how or if it's supported. AFAIK the overlay is a copy of the syscfg partition which is a persistent nand storage location.

rip057, 2015-02-10:
I got

alirz1, 2015-02-10:
@chadster.
Thanks . Although I thought mcwrt was your creation and that would know it inside out.

@rip057 you got what?

rip057, 2015-02-10:
That was a mistaken pocket email if you would believe that, but I had pivot
overlay working, but have since bricked the wrt1900ac, and dont have my
config file to look at

closed 2015-08-17 by Chadster766