Chadster766/McWRT #88: V1.08 wifi unable to connect

rashmac, 2015-02-09:
I have had  OpenWrt Chaos Calmer r43823 from lifehacksback - http://wiki.openwrt.org/toh/linksys/wrt1900ac . running for a while now but have experienced wireless drop issues . therefore I wanted to go back to the Beloved MCWRT. but I noticed and remembered the reason why I moved away from it.

No matter what I do I cannot connect ,. windows / smart phones and macbook to the wireless network 
I have tried changing the keys setting different encryption techniques . etc .. but no matter what I do the following response if received from my devices
- iphone - unable to connect
- Windows 7 laptop - unable to connect
- macbook air - invalid password 

I  can't be the only person having this issue , can I ? lol
I am sure its something silly , but would you know what ? I am currently back to lifehacksback's build as I need a working internet connection at home . so I can't test anything immediately . was just wondering if you have come across this and perhaps have a fix I could try at a later date

Thx

speedandy, 2015-02-09:
I have used this version with my Toshiba laptop (5Ghz n), my wife's MacBook air (ac), 3 iphones (2 ac, 1n), 2 ipads (1 ac, 1 n) and 2 ipods (n), my apple TV etc.  with no problems.  Have you checked channels for interference? It may be that simple.

Otherwise check your encryption is WPA2 for n and ac connections, and you are using 20 Mhz of spectrum, not 40 for 2.4 Ghz devices and 40 not 80 for 5 Ghz devices.  Apple devices hate anything but the former in my experience.

Chadster766, 2015-02-09:
The default wireless settings of the AAv1.0.8 image work with those devices and is secured by WPA2. Only changing the wireless SSID's and passwords is typically required.

Settings from different major versions of OpenWRT are not compatible. You might have to clear the previous settings, reboot and reconfigure.

```
rm -rf /overlay/*
reboot
```
Did you compile your image from source or are you using a the release image download?

rashmac, 2015-02-09:
hmm Chaster you might be right here , I downloaded the image from here . I did not compile it .
however I noticed that it automatically had set my SSID just like the previous firmware I had .. is there is shell command to reset it ? or shall I just reset it using the firmware after I have flashed it with Mcwrt 

I am running Chaos Calmer r43823 again as mentioned previously .. I will test Mcwrt again this weekend if I get a chance 

Thanks

Chadster766, 2015-02-09:
After you load McWRT you can clear the overlay folder as shown above which should reboot the router with AAv1.0.8 defaults.

rip057, 2015-02-09:
I think you have to use wpa2 personal and force aes/ccmp encryption

Some devices dont work well changing algorithms

beetee2, 2015-02-26:
Clear the overlay folder? That is obviously a command but how do I issue a command to the wrt1900ac? Do I seriously have to buy an additional cable and run linux to be able to use any open source firmware for this router?

Chadster766, 2015-02-26:
The McWRT wireless driver doesn't supply the UI with wireless statistics. You can get that from connected client devices. 

A USB to TTL cable is required if you need to recover from a bad firmware load. They are approx $10.00

OpenWRT is a Linux system like most other embedded devices. 

OpenWRT allows for SSH access by default once you change the default router password. 

A good SSH software for Windows is Putty.

beetee2, 2015-02-26:
Thanks for the info, I hadn't considered using ssh from Windows!

Can you change the SSIDs that McWRT creates? Because that's literally the
first thing I did after flashing yesterday, and then I tested, and neither
channel worked.

Changed the SSIDs, added passwords with WPA2 aes encryption, didn't work,
fiddled with settings for an hour, nothing ever worked.

Chadster766, 2015-02-26:
The wireless defaults are a known good. WPA2 Personal is the default wireless security and belkin123 the default wireless password. 

I recommend changing only the SSID's and passwords to start. 

Clear the overlay folder or reload the firmware with keep settings unchecked to restore default settings.

rip057, 2015-02-26:
Im just gonna be honest, and please dont take it totally the wrong way, but
maybe open source firmware isnt for you... or at least using this 250
retailed price device as a learning block isnt for you... you have the real
possibility of permanently bricking devices setting up known good builds on
devices and everything for the wrt1900ac is experimental or what is known
as bleeding edge from what I know...

A working knowledge of linux is not required on all open source firmwares,
as it has been gui'd to some degree, but is almost essential to getting
anything done with the wrt1900ac...

That being said if you have money go for it, but get a book on basic bash
commands it will help you out

The wireless defaults are a known good. WPA2 Personal is the default
wireless security and belkin123 the default wireless password.

I recommend changing only the SSID's and passwords to start.

Clear the overlay folder or reload the firmware with keep settings
unchecked to restore default settings.

beetee2, 2015-02-26:
Thanks for your concern, but I've been a software developer for 10 years. I've been running Linux on at least one machine in my house since Ubuntu 7. I've published Android apps and I'm a professional .Net developer. My question wasn't one of technical ineptitude, but rather one of confusion about the level of effort required in order to run custom firmware on this particular device. There isn't a single place for detailed troubleshooting / flashing information that I have found, for example I had to comment on this github bug in order to find out I needed to remove the overlay.

rip057, 2015-02-26:
All you need to do to remove the overlay is ssh or telnet in and
rm -Rf /overlay

Sorry for coming off half cocked there is a great thread in the discussion
section of openwrt just search wrt1900ac its the 90 some od page thread
last I checked

Just in that thread there are a ton of noobs (along w/ some seasoned vets)
and I always warn people cause it needs to be done this new firmware isnt
for the novice, I bricked my own and just havent gotten around to the ttl
cabe cause there is one glitch with the iptables string function not
working that makes the router kind of useless for me right now...

beetee2, 2015-02-26:
I understand, no offense taken, this isn't my first day on the internet. =)

closed 2015-08-17 by Chadster766