Chadster766/McWRT #90: Add DNScrypt-proxy package

AliceGrey, 2015-02-25:
I would really love to have dnscrypt on the wrt. It's a really popular package to use.

Chadster766, 2015-02-25:
Did you try to compile it? If you need help I can advise.

AliceGrey, 2015-02-25:
Compile dnscrypt? Or McWrt?

Chadster766, 2015-02-25:
You can compile both or just the package.

http://wiki.openwrt.org/doc/howtobuild/single.package

AliceGrey, 2015-02-25:
Which architecture is the wrt?

Chadster766, 2015-02-25:
If you clone the McWRT repo and run "update.sh" and then "make menuconfig"; the arch will be listed and selected.

You can build OpenWRT on x32 or x64 Linux system.

AliceGrey, 2015-02-25:
I meant out of these architectures. http://downloads.openwrt.org/attitude_adjustment/12.09/ Which one do I build dnscrypt for?

Chadster766, 2015-02-25:
It's not in that list.

There are two options:
AA = armadaxp
BB\CC = mvebu

Chadster766, 2015-02-25:
To build this package for the McWRT firmware. You clone this repo. Then you can add the package source and in "make menuconfig" add the package to the build. You will most likely have to troubleshoot compile errors and find or make patches.

closed 2015-08-17 by Chadster766