Chadster766/McWRT #91: WRT1900AC WAN goes up down

alirz1, 2015-03-07:
MCWRT: OpenWrt Attitude Adjustment 12.09.1 / LuCI Trunk (svn-r10537)
Router: WRT1900AC

i've been running MCWRT on the router for 3 months. Never really noticed the issue till i came across the system logs and see that the wan connection seems to be restarted many times a day..

I've masked my wan Ip in the log below.  Any ideas why this would be happening. This seems to be recover pretty quickly so perhaps thats why i haven't noticed any issues with my internet access but would explain why maybe sometimes my ps4, ps3 signs out in the middle of a game. Otherwise its hard to catch this issue.
Note that i have a cable modem connection. My IP doesnt really change often, perhaps once or twice a year.....

Mar  6 16:37:42 OpenWrt daemon.err miniupnpd[1711]: ioctl(s, SIOCGIFADDR, ...): Cannot assign requested address
Mar  6 16:37:42 OpenWrt daemon.err miniupnpd[1711]: Failed to get IP for interface eth1
Mar  6 16:37:42 OpenWrt daemon.warn miniupnpd[1711]: SendNATPMPPublicAddressChangeNotification: cannot get public IP address, stopping
Mar  6 16:37:42 OpenWrt daemon.notice netifd: Interface 'wan6' is now down
Mar  6 16:37:42 OpenWrt user.notice root: stopping ntpclient
Mar  6 16:37:45 OpenWrt daemon.notice netifd: wan (1378): Sending discover...
Mar  6 16:37:45 OpenWrt daemon.warn dnsmasq[2775]: no servers found in /tmp/resolv.conf.auto, will retry
Mar  6 16:37:45 OpenWrt daemon.info dnsmasq-dhcp[2775]: DHCPDISCOVER(br-lan) 6c:ad:f8:18:2e:a0 
Mar  6 16:37:45 OpenWrt daemon.info dnsmasq-dhcp[2775]: DHCPOFFER(br-lan) 192.168.1.40 6c:ad:f8:18:2e:a0 
Mar  6 16:37:48 OpenWrt daemon.notice netifd: wan (1378): Sending discover...
Mar  6 16:37:48 OpenWrt daemon.notice netifd: wan (1378): Sending select for 184.X.X.X...
Mar  6 16:37:48 OpenWrt daemon.info dnsmasq-dhcp[2775]: DHCPDISCOVER(br-lan) 6c:ad:f8:18:2e:a0 
Mar  6 16:37:48 OpenWrt daemon.info dnsmasq-dhcp[2775]: DHCPOFFER(br-lan) 192.168.1.40 6c:ad:f8:18:2e:a0 
Mar  6 16:37:48 OpenWrt daemon.notice netifd: wan (1378): Lease of 184.X.X.X obtained, lease time 3600
Mar  6 16:37:48 OpenWrt daemon.notice netifd: Interface 'wan' is now up
Mar  6 16:37:48 OpenWrt daemon.notice odhcp6c[7378]: (re)starting transaction on eth1
Mar  6 16:37:48 OpenWrt daemon.notice odhcp6c[7378]: Sending SOLICIT (timeout 4294967295s)

alirz1, 2015-03-07:
Update, I've been able to determine that this is happening every 30 minutes exactly...there must be something on the router doing some check every 30 min some place.
I'm wondering if this is just a logging issue and perhaps the wan connection is not being rest? I'll have to monitor the connection right at the time when I'm expecting this error message to be logged

Chadster766, 2015-03-07:
One is WAN6 which is IPv6 which I don't think is required by the log. You can safely disable WAN6 startup. 

Your ISP is giving you an ip addresses that expires every 3600s or 1 hour.

alirz1, 2015-03-07:
Thanks I will stop wan6 from comin up on boot.
As for the up and down, so that doesn't mean that the Internet connection going up and down every time I see that message in the logs? It's just lease renew notification?

Chadster766, 2015-03-07:
Yes I think so

alirz1, 2015-03-07:
You know what, i can confirm that internet connection does go down when that message occurs.
I was playing an online game on PS3, when suddenly it signed me out of the PSN network.
i ran to check the router log and behold, i saw the same thing below.

Even if my ISP lease is for 3600 min, as i said before this is happening every 30 minutes. 23 min and 53 minute mark of each hour.
Is there a way i can prolong this check, or disable it altogether..

Mar  7 11:53:56 OpenWrt daemon.notice netifd: wan (1378): Received DHCP NAK
Mar  7 11:53:56 OpenWrt daemon.notice netifd: Interface 'wan' has lost the connection
Mar  7 11:53:56 OpenWrt daemon.warn dnsmasq[4773]: no servers found in /tmp/resolv.conf.auto, will retry
Mar  7 11:53:56 OpenWrt daemon.err miniupnpd[7962]: ioctl(s, SIOCGIFADDR, ...): Cannot assign requested address
Mar  7 11:53:56 OpenWrt daemon.err miniupnpd[7962]: Failed to get IP for interface eth1
Mar  7 11:53:56 OpenWrt daemon.warn miniupnpd[7962]: SendNATPMPPublicAddressChangeNotification: cannot get public IP address, stopping
Mar  7 11:53:56 OpenWrt user.notice root: stopping ntpclient
Mar  7 11:53:58 OpenWrt daemon.info hostapd: wdev0ap0: STA 6c:ad:f8:18:2e:a0 WPA: group key handshake completed (RSN)
Mar  7 11:54:01 OpenWrt user.warn igmpproxy[3789]: The origin for route 239.255.255.250 changed from 192.168.1.4 to 192.168.1.14
Mar  7 11:54:02 OpenWrt user.warn igmpproxy[3789]: The origin for route 239.255.255.250 changed from 192.168.1.14 to 192.168.1.4
Mar  7 11:54:02 OpenWrt daemon.notice netifd: wan (1378): Sending select for 184..X.X.X...
Mar  7 11:54:02 OpenWrt daemon.notice netifd: wan (1378): Lease of 184.X.X.X obtained, lease time 3600
Mar  7 11:54:02 OpenWrt daemon.notice netifd: Interface 'wan' is now up

alirz1, 2015-03-07:
And again: notice the timestamps

Mar  7 12:24:02 OpenWrt daemon.notice netifd: Interface 'wan' has lost the connection
Mar  7 12:24:02 OpenWrt daemon.warn dnsmasq[15744]: no servers found in /tmp/resolv.conf.auto, will retry
Mar  7 12:24:02 OpenWrt daemon.err miniupnpd[12029]: ioctl(s, SIOCGIFADDR, ...): Cannot assign requested address
Mar  7 12:24:02 OpenWrt daemon.err miniupnpd[12029]: Failed to get IP for interface eth1
Mar  7 12:24:02 OpenWrt daemon.warn miniupnpd[12029]: SendNATPMPPublicAddressChangeNotification: cannot get public IP address, stopping
Mar  7 12:24:02 OpenWrt daemon.err miniupnpd[12029]: ioctl(s, SIOCGIFADDR, ...): Cannot assign requested address
Mar  7 12:24:02 OpenWrt daemon.err miniupnpd[12029]: ioctl(s, SIOCGIFADDR, ...): Cannot assign requested address
Mar  7 12:24:03 OpenWrt user.notice root: stopping ntpclient
Mar  7 12:24:05 OpenWrt daemon.info dnsmasq-dhcp[15744]: DHCPINFORM(br-lan) 192.168.1.2 e0:69:95:da:d7:e1 
Mar  7 12:24:05 OpenWrt daemon.info dnsmasq-dhcp[15744]: DHCPACK(br-lan) 192.168.1.2 e0:69:95:da:d7:e1 desktop
Mar  7 12:24:09 OpenWrt daemon.notice netifd: wan (1378): Sending select for 184.163.182.31...
Mar  7 12:24:09 OpenWrt daemon.notice netifd: wan (1378): Lease of 184.163.182.31 obtained, lease time 3600
Mar  7 12:24:09 OpenWrt daemon.notice netifd: Interface 'wan' is now up

alirz1, 2015-03-07:
Mar  7 18:42:09 OpenWrt daemon.notice netifd: wan (1379): Sending renew...
Mar  7 18:42:09 OpenWrt daemon.notice netifd: wan (1379): Received DHCP NAK
Mar  7 18:42:10 OpenWrt daemon.notice netifd: Interface 'wan' has lost the connection

I think im having the same issue detailed in this old ticket. based on that thread, it should be fixed in my AA version

https://dev.openwrt.org/ticket/12114

Chadster766, 2015-03-08:
Thanks for the link. It isn't clear to me in the ticket if the fix actually worked for the user or not since Jow marked it as "works for me". Maybe you could hunt down a patch for this issue.

alirz1, 2015-03-08:
While i think of myself as an excellent troubleshooter and IT is my field... i dont know much coding and GIT stuff. I've been looking for more info on this but haven't really found anything except someone mentioning that by setting the WAN interface to static IP, seemed to resolve the issue for them
Even though my IP doesnt change and i consider it pretty much static i dont know what side effects setting the WAN port ot use static protocol has.

I really like MCWRT and dont want to switch to CC, as they have other serious issues that i dont want to get into.

alirz1, 2015-03-08:
What about this looks like good info on this issue

https://dev.openwrt.org/ticket/14144   (though this is for PPOE connections and this looks more severe)
and
https://dev.openwrt.org/ticket/138

alirz1, 2015-03-16:
Is there a way to disable dhcp renew at the 50% of lease time remaining. I believe that is the dhcp standard.
Currently my router is unable to renew the lease with my ISP. I spoke to them but no avail.
Here's the scenario...
I get a 24hr dhcp lease from my ISP. My IP doesn't really change .
After 12hrs , the router tries to renew the lease so sends a renew request to the DHCP SETVER OF THE ISP.
the server sends a DHCP NAK back, which is a deny.
The router sees that as something wrong and restarts the wan interface.

I'm trying to prolong this reoccurrence rather than trying to fix it now. Is there a way the router tries to renew only after the 24hr lease is up and before
Thanks

Chadster766, 2015-03-16:
Does this still occur if you stop the wan_monitor?

Command:
killall wan_monitor

alirz1, 2015-03-16:
I didn't know about that. Will give it a shot..thanks

alirz1, 2015-03-16:
looks like it happened despite having killed the wan-monitor process

Mar 16 12:05:38 OpenWrt daemon.notice netifd: wan (1381): Sending renew...
Mar 16 12:05:38 OpenWrt daemon.notice netifd: wan (1381): Received DHCP NAK
Mar 16 12:05:38 OpenWrt daemon.notice netifd: Interface 'wan' has lost the connection
Mar 16 12:05:38 OpenWrt daemon.err miniupnpd[20010]: ioctl(s, SIOCGIFADDR, ...): Cannot assign requested address
Mar 16 12:05:38 OpenWrt daemon.err miniupnpd[20010]: Failed to get IP for interface eth1
Mar 16 12:05:38 OpenWrt daemon.warn miniupnpd[20010]: SendNATPMPPublicAddressChangeNotification: cannot get public IP address, stopping
Mar 16 12:05:40 OpenWrt daemon.warn dnsmasq[2738]: no servers found in /tmp/resolv.conf.auto, will retry
Mar 16 12:05:44 OpenWrt daemon.notice netifd: wan (1381): Sending select for 173.176.X.X...
Mar 16 12:05:44 OpenWrt daemon.notice netifd: wan (1381): Lease of 173.176.131.X.X obtained, lease time 86400

Chadster766, 2015-03-16:
Actually that's a relief :smile:

alirz1, 2015-03-16:
Any other thoughts on how to delay or perhaps even disable the dhcp renew requests?

alirz1, 2015-03-22:
Just wanted to add that I decided to try the CC builds. Currently running the latest from Kaloz dated march 20th and this wan up and down does NOT happen with the firmware..
At the time of the lease renew, the router simply renews the lease without restarting the WAN interface.

So something in MCWRT has that behavior. Perhaps Udhcpc ?

Chadster766, 2015-08-18:
I've started working on this issue.

Chadster766, 2015-08-21:
I tested with MCWRT 2.1 and the DHCP NAK behaves the same as my computer so its working as it should :smile:

closed 2015-08-21 by Chadster766