Chadster766/McWRT #92: No further development on MCWRT?

alirz1, 2015-03-09:
Is it correct that there will be no further development and fixes for MCWRT?

Thanks

Chadster766, 2015-03-09:
I won't be doing anymore work on McWRT since CC should be released soon. I'm still accepting Pull Requests for code updates for McWRT but no one is doing them.

Right now I'm running the latest Linksys firmware and taking advantage of the Smart Wifi capabilities. It's working well and provides the features I need for the most part. The main issue I have with the Linksys firmware is the DNS Proxy doesn't allow for OpenDNS filtering.

alirz1, 2015-03-09:
ok :(
CC has too many issues, biggest being wireless issues. Not to mention router freezes that everyone seems to be talking about.

OEM firmware doesnt do it for me. I need to have Broadcast network support for wake on WAN for.
I also run some custom scripts/cronjobs on the router that sends me emails if my wan ip changes etc..

Does anyone else actively @mojolacerator
I have that TTL cables also. How do you know, which color wire plugs on which serial pin of the router?
Thanks for all your help.

raversnet, 2015-03-09:
Chad, so you will continue the work on this router once CC is released
then?  Or is it expected to support linksys 1900ac by default?

Ive been using AA 1.0.8

Robert McCallum
1532 Victoria Ave.
Regina, SK
S4P 0P5

M - 306-216-0526
W - 306-787-2941

Aerogeek, 2015-03-09:
Sorry to hear that Chad. Great effort though on all you've achieved.

Whats 'CC'? (Link?)

Cheers, Pete

ellismichaelc, 2015-03-09:
I recently flashed the latest CC and a couple other builds in the OpenWRT thread for the WRT1900AC.. I couldn’t get ANY of them stable, especially the Wi-Fi as stated. I flashed back to McWRT and it was perfect again.

raversnet, 2015-03-10:
I did a search and realized support had finally started for WRT1900AC.  

Thanks so much Chad for all your work.  Based on what i've read about the CC builds i think i'll stick with McWRT and its stability for the time being.  

Cheers to you Chad

closed 2015-08-17 by Chadster766