Chadster766/McWRT #93: Sonos Speakers

OllieB, 2015-03-16:
Got this weird issue with my wireless Sonos speakers and the WRT1900AC. Initially, they work juts fine, but after a while they stop responding. It looks like something weird is going on with the ARP table:

? (10.0.0.201) at (incomplete) on en0 ifscope [ethernet]
? (10.0.0.230) at 0:e:58:6b:ba:9e on en0 ifscope [ethernet]

Even the one that it has a MAC for, it isn't a complete MAC address, weird huh?

I have another speaker connected to a seperate wireless router (same SSID), but that one works just fine...

Any thoughts as to what I might be fighting here? Thanks!

Chadster766, 2015-03-16:
What OpenWRT firmware are you running? The reason I ask is the adapter seems to be labeled differently than McWRT.

OllieB, 2015-03-16:
That's actually the output from my mac, I haven't tried checking the arp table on the router, that would be a good idea. It's the latest McWRT, 1.0.8, which has been splendid, thank you!

Chadster766, 2015-03-16:
I you are going to check the arp table on the router the below links might be helpful:

https://forum.openwrt.org/viewtopic.php?id=6187

OllieB, 2015-03-16:
So I've been looking at this for a bit now, and I'm finding that with two devices on the same wireless network (my Mac and iPhone), one is able to ping the speakers, but the other isn't. Doing a tcpdump on the router shows the ping request never actually hits br-lan from the iPhone. The iPhone can ping all other connected hosts. Any ideas what might be going on or what I can do to troubleshoot further?

Thanks :)

closed 2015-08-17 by Chadster766