Chadster766/McWRT #94: TTL Pinout (maybe) Incorrectly

norpol, 2015-03-18:
Hi,
some friend of mine tried to gain access to the bootloader via. the 3.3 volt serial connection.
He didn't succeeded until we checked the pins instead of the USB to serial adapter with two different oscilloscopes. Both displayed when connected to first pin (GND) and the 4th pin a 'pull' to 3.3 volts and then some signals, but they were going up to ~3.5 volts instead of down to 0 volts again. After checking all other things (using different ground etc) we figured out that the 2nd pin is TX and the 4th is RX. According to one of the pictures in the [wiki](https://github.com/Chadster766/McWRT/wiki/Linksys-WRT1900AC-Serial-Port) TX is the 4th and RX the 2nd pin, so you've maybe swapped things. PS: [Some other](https://github.com/Chadster766/McWRT/wiki/WRT1900ac---Gaining-accesss-to-the-UBOOT-console-via-the-TTL-port#creating-a-cable-to-connect-to-the-ttl-port) wiki page is telling us the correct pinout, but is warning us with a "work in progress" headline.

Anyway, thanks for doing all of your work. I hope I could describe everything correctly, I can correspond with a short time-delay, so ask me anything.

Chadster766, 2015-03-19:
Thanks for the notice. I've reversed the Rx and Tx on the Wiki page you mentioned.

norpol, 2015-03-20:
I think now you've got both wrong. According to our tests with the oscilloscope (and later with the TTY-USB connector) the correct pinout  is GND[1], TX[2] , RX[4].

Chadster766, 2015-03-20:
Just for confirmation you used this for your electrical testing 0VDC = 1 in TTL vs RS232 +5 - 12VDC = 1.

norpol, 2015-03-20:
Oh, wait I'm wrong. Another friend of mine just told me what we were doing wrong. If you're talking about RX you mean receive on the chip that is located on the board. So if I connect with my tty-adapter I need to connect RX with TX & TX with RX. So the wiki entries are correct, we just didn't understand your label, since we were to inexperienced.

Thanks for your time. :) You can close the issue...

Chadster766, 2015-03-20:
You were right about the one Wiki page being wrong though :smile:

closed 2015-03-20 by Chadster766