Chadster766/McWRT #95: kwboot.tar.gz link is 0 byte file

rip057, 2015-03-24:
just what it says... the folder linked in the uboot recovery wiki has a 0 byte file for kwboot.tar.gz and fedora 21 cant run the precompiled kwboot... plz fix i really need to redo the bootloader  i can load stock but every version of openwrt causes a system halted

Chadster766, 2015-03-24:
On my webserver I couldn't put the compressed file no matter what the format. I've put the source files instead.

Chadster766, 2015-03-24:
Nope the *.c source files are not uploading. I'm discussing with my Host service.

You can download from here: http://filebin.ca/1hAJVQqY1xFN/kwboot.tar.bz2

closed 2015-03-24 by Chadster766

rip057, 2015-03-24:
hey thanks alot, im sure your super busy, so I really appreciate the expediency

rip057, 2015-03-24:
i know this is nit picky and anyone with a brain should be able to figure this out, but gcc-4.8.3 (the one i use) outputs the compilation to a.out not kwboot ... very simple but might screw someone up