Chadster766/McWRT #96: Possibly a stupid question... but does this work for EA6900? or just WRT1900AC?

joshuairl, 2015-03-24:


Chadster766, 2015-03-24:
Just the WRT1900AC

joshuairl, 2015-03-24:
No way to run OpenWRT on EA6900 that you know of?

joshuairl, 2015-03-24:
Thanks for quick response too!

Chadster766, 2015-03-24:
Sorry I not AFAIK but I've heard of DD-WRT being loaded on the EA6900. I recommend you post in the OpenWRT forum because during development I thought I saw support being explorered for other EA series routers.

closed 2015-03-24 by Chadster766

alirz1, 2015-03-24:
I have used DD wet on the 6900 about 4 months ago. It is supported by ddwrt

joshuairl, 2015-03-24:
I've tried various builds close to a year ago and weren't stable but maybe it's improved

Chadster766, 2015-03-24:
Finally someone that confirms it's possible for DD-WRT on the EA6900. I would be curious on how the newer version of DD-WRT work on the EA6900 since I have one of those as well. I find the EA6900 very fast wired and wireless. The new Linksys firmware for the EA6900 is good too.

joshuairl, 2015-03-24:
I personally hate how simple the smart wifi UI is and it feels very limited.

Chadster766, 2015-03-24:
Yes compared to OpenWRT and DD-WRT the Linksys firmware doesn't have nearly the same amount of features.

For those with basic network needs than the Linksys firmware is a very good option.

alirz1, 2015-03-24:
My Primary reason for using dd-wrt or openwrt is the fact that it lets me use Wake on WAN using broadcast Ips on the LAN. Linksys oem firmwares dont allow that..But i also have other scripts and 3rd party apps running on my router now thanks to these open firmwares.
I had been using dd-wrt on various routers for the past 6 or so years.
Only jumped on the openwrt wagon 3 months ago when i got the WRT1900AC..Used Chadster build  uptil last week without any issues till i found an issue with the WAN interface restarting at lease renew time....Decided to try the lastest CC builds and so far they have been good.

Koulio, 2016-02-01:
http://www.linksysinfo.org/index.php?threads/asuswrt-merlin-tomato-shibby-or-toastman-dd-wrt-on-linksys-ea6900.71718/

Chadster766, 2016-02-01:
Sorry McWRT is only compatible with the WRT1900AC V1.

I've heard of DD-WRT running on the EA6900 but it doesn't seem to be officially supported.

kyshop38, 2020-08-08:
Try ea6900 merlin

rip057, 2020-08-08:
Nice Necro.  It brings me back :)