Chadster766/McWRT #98: QOS still broken - relies on fq_codel

tusc, 2015-08-24:
It doesn't appear as though this has been looked at. If you enable QOS within luci you'll note your downstream speed is cut down to a 1/4 of the original speed. The reason for this is that the script does not completely succeed

If you run the command in the script /usr/bin/qos-start:

/usr/lib/qos/generate.sh all | sh

It will respond with the following:

RTNETLINK answers: No such file or directory
RTNETLINK answers: No such file or directory
RTNETLINK answers: No such file or directory
RTNETLINK answers: No such file or directory
RTNETLINK answers: No such file or directory
RTNETLINK answers: No such file or directory
RTNETLINK answers: No such file or directory
RTNETLINK answers: No such file or directory

If you rerun the command like this with debugging:
/usr/lib/qos/generate.sh all | sh -x

it will show the lines it is complaining about: fq_codel.

The codel algorithm is heavily depended on within qos-scripts. It is a feature that is available with linux kernel 3.6 or greater. Since McWRT is running a 3.2 kernel this won't work.

Chadster766, 2015-08-24:
Maybe an older OpenWRT QOS script would work better. From my testing it does work but only if you are under 7mbps upstream and downstream maximums.

tusc, 2015-08-24:
Ah, that might explain it. I'm on Comcast with 80Mbps down and 10Mbps up. When the script is executed my downstream is kept at 14Mbps.

I've tried looking for older scripts but they all seem to depend on fq_codel.

Chadster766, 2015-08-24:
Can fq_codel be added to the kernel in a patch or as a kernel module?

tusc, 2015-08-24:
Don't believe so.

From http://www.bufferbloat.net/projects/codel/wiki

"CoDel - in order to run well at line rate - requires the Linux 3.3 Byte Queue Limits . It has proven too hard to backport BQL to Linux 3.2 or earlier (an attempt for 3.2 exists, but no driver support), so you will need to upgrade to Linux 3.5 or later, and have a driver that supports BQL (only about 24 as of the present writing)."

Chadster766, 2015-08-24:
Thanks for the CoDel info.

It looks like the existing script gets counter values directly from the network driver. I wish I knew more about that stuff but the script will probably need to be tweaked for the new higher counter values.

closed 2020-02-14 by Chadster766