Chadster766/McWRT #99: cannot change/set password

LogicoZone, 2015-08-26:
I checked out the latest source code and do a compilation. 

I telnet in and got this error when running "passwd" for root.
passwd: /etc/passwd: Read-only file system
passwd: can't update password file /etc/passwd
when run "df", here is the output.
root@McWRT:/# df
Filesystem           1K-blocks      Used Available Use% Mounted on
rootfs                   37888     13684     24204  36% /
/dev/root                37888     13684     24204  36% /rom
devtmpfs                126560         0    126560   0% /dev
tmpfs                   126672       756    125916   1% /tmp
root                    126672        84    126588   0% /tmp/root
overlayfs:/tmp/root     126672        84    126588   0% /tmp/root
overlayfs:/overlay       37888     13684     24204  36% /

From a working McWRT image (v2.3), the output of df is like this:

root@McWRT:/etc# df
Filesystem           1K-blocks      Used Available Use% Mounted on
rootfs                   31424       312     29476   1% /
/dev/root                37888     23240     14648  61% /rom
devtmpfs                126560         0    126560   0% /dev
tmpfs                   126672       480    126192   0% /tmp
ubi0:syscfg              31424       312     29476   1% /tmp/syscfg
ubi0:syscfg              31424       312     29476   1% /overlay
overlayfs:/overlay       31424       312     29476   1% /

Does anyone know what I missed from the configuration of McWRT?

Chadster766, 2015-08-26:
I don't think you missed anything it's more likely your overlay has an issue.

First try to delete the overlay with the below command and reboot/reconfigure:
rm -rf /overlay

or 

rm -rf /overlay/*

LogicoZone, 2015-08-28:
This doesn't work. It seems the mount point is incorrect. I don't have this from the image you compiled. 

ubi0:syscfg 31424 312 29476 1% /tmp/syscfg
ubi0:syscfg 31424 312 29476 1% /overlay

Here is when I tried to delete the overlay dir and it failed. I also tried to delete the contents within overlay and reboot and still same issue. 

root@McWRT:/# passwd
Changing password for root
New password: 
Bad password: too weak
Retype password: 
passwd: /etc/passwd: Read-only file system
passwd: can't update password file /etc/passwd
root@McWRT:/# rm -rf /overlay/
rm: can't remove '/overlay': Read-only file system
root@McWRT:/# rm -rf /overlay
rm: can't remove '/overlay': Read-only file system
root@McWRT:/#

Chadster766, 2015-08-28:
Try cloning the repo and doing a "git pull" just in case. Then you will have to do a "scripts/feeds update" to bring the feed packages in.

LogicoZone, 2015-08-28:
I did do clone and pull then update the package. I'll start it over again. In the meantime, it seems I am not able to flash OpenWRT firmware. I tried from GUI and it looks fine and reboot but I'm still at McWRT. Any comment?

Chadster766, 2015-08-28:
Do you have the USB to TTL console connection?

Also you can flash the V3.2 image with "keep settings" unchecked and then flash any firmware AFAIK.

LogicoZone, 2015-08-28:
I'll try again. I did use 2.3 image and flash openwrt. I'll use usb to ttl later.

Chadster766, 2015-08-28:
Worse case would be to clean the MTD partitions in U-boot:
https://github.com/Chadster766/McWRT/wiki/Clean-MTD-Partitions

Chadster766, 2015-08-31:
I'm assuming you figure this out.

closed 2015-08-31 by Chadster766