Chadster766/McWRT #100: Can't route ipv6 packets from LAN clients

bikeNomad, 2015-09-09:
Hello,

Thanks for your work on McWRT; it looks great!

I'm trying to get ipv6 to work correctly with my newly-installed McWRT firmware on my WRT1900AC (v1).

So far, it's working somewhat better than on the OpenWRT BB rc3 that I'd had on the router (that is, the router has an IPV6 address and knows its prefix delegation).

I have an ISP (Comcast) that provides a native dual-stack environment.

On the factory firmware (or connected directly to the cable modem), I get proper ipv6 operation from my client machines (which are mostly Macs and Linux NAS boxes).

However, on McWRT I don't get ipv6 packets routed to/from my client machines, despite them getting ipv6 addresses (2601:602:8000:7eb8::/64) from the DHCP server

I can "ping -6" from the router to the Internet (as well as get IPV6 DNS resolution).

I can't ping from the router to the client machines' IPV6 addresses:

    root@Metamagix:~# ping -6 2601:602:8000:7eb8::f53
    PING 2601:602:8000:7eb8::f53 (2601:602:8000:7eb8::f53): 56 data bytes
    ^C
    --- 2601:602:8000:7eb8::f53 ping statistics ---
    4 packets transmitted, 0 packets received, 100% packet loss
    root@Metamagix:~# ping -6 -I br-lan 2601:602:8000:7eb8::f53
    PING 2601:602:8000:7eb8::f53 (2601:602:8000:7eb8::f53): 56 data bytes
    ^C
    --- 2601:602:8000:7eb8::f53 ping statistics ---
    8 packets transmitted, 0 packets received, 100% packet loss
    root@Metamagix:~# ping -6 -I eth0 2601:602:8000:7eb8::f53
    PING 2601:602:8000:7eb8::f53 (2601:602:8000:7eb8::f53): 56 data bytes
    ping: sendto: Network is unreachable

ifconfig reports:

    br-lan    Link encap:Ethernet  HWaddr 94:10:3E:14:89:30
              inet addr:192.168.1.1  Bcast:192.168.1.255  Mask:255.255.255.0
              inet6 addr: 2001:558:600a:57:5853:c33:9f54:2ddd/128 Scope:Global
              inet6 addr: fe80::9610:3eff:fe14:8930/64 Scope:Link
              inet6 addr: 2601:602:8000:7eb8::1/64 Scope:Global
              UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
              RX packets:2846036 errors:0 dropped:0 overruns:0 frame:0
              TX packets:2878172 errors:0 dropped:0 overruns:0 carrier:0
              collisions:0 txqueuelen:0
              RX bytes:342494332 (326.6 MiB)  TX bytes:2668402403 (2.4 GiB)
    
    eth0      Link encap:Ethernet  HWaddr 94:10:3E:14:89:30
              UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
              RX packets:10472718 errors:0 dropped:0 overruns:0 frame:0
              TX packets:15365155 errors:0 dropped:0 overruns:0 carrier:0
              collisions:0 txqueuelen:1000
              RX bytes:812530486 (774.8 MiB)  TX bytes:223325238 (212.9 MiB)
              Interrupt:8
    
    eth1      Link encap:Ethernet  HWaddr 00:50:43:02:00:00
              inet addr:73.157.91.74  Bcast:73.157.91.255  Mask:255.255.254.0
              inet6 addr: 2001:558:600a:57:5853:c33:9f54:2ddd/128 Scope:Global
              inet6 addr: fe80::250:43ff:fe02:0/64 Scope:Link
              UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
              RX packets:2744010 errors:0 dropped:0 overruns:0 frame:0
              TX packets:2144308 errors:0 dropped:0 overruns:0 carrier:0
              collisions:0 txqueuelen:1000
              RX bytes:2611859661 (2.4 GiB)  TX bytes:322963091 (308.0 MiB)
              Interrupt:10

I haven't touched the firewall configuration.

I'm getting lots of messages in the syslog like this:

    2015-09-09T08:53:23-07:00 192.168.1.1 6relayd[2296]: Failed to relay to ff02::1%br-lan (Network is unreachable)
    2015-09-09T08:53:24-07:00 192.168.1.1 dnsmasq[3580]: read /etc/hosts - 1 addresses
    2015-09-09T08:53:24-07:00 192.168.1.1 dnsmasq[3580]: read /tmp/hosts/6relayd - 0 addresses
    2015-09-09T08:53:24-07:00 192.168.1.1 dnsmasq-dhcp[3580]: read /etc/ethers - 0 addresses
    2015-09-09T08:53:26-07:00 192.168.1.1 6relayd[2296]: Failed to relay to ff02::1%br-lan (Network is unreachable)
    2015-09-09T08:53:29-07:00 192.168.1.1 6relayd[2296]: Failed to relay to ff02::1%br-lan (Network is unreachable)
    2015-09-09T08:53:32-07:00 192.168.1.1 6relayd[2296]: Failed to relay to ff02::1%br-lan (Network is unreachable)
    2015-09-09T08:53:35-07:00 192.168.1.1 6relayd[2296]: Failed to relay to ff02::1%br-lan (Network is unreachable)
    2015-09-09T08:53:38-07:00 192.168.1.1 6relayd[2296]: Failed to relay to ff02::1%br-lan (Network is unreachable)
    2015-09-09T08:53:41-07:00 192.168.1.1 6relayd[2296]: Failed to relay to ff02::1%br-lan (Network is unreachable)
    2015-09-09T08:53:44-07:00 192.168.1.1 dnsmasq[3580]: read /etc/hosts - 1 addresses
    2015-09-09T08:53:44-07:00 192.168.1.1 dnsmasq[3580]: read /tmp/hosts/6relayd - 2 addresses

My config files look like this:

    root@Metamagix:~# cat /etc/config/6relayd

    config server 'default'
      option master 'wan6'
      list network 'lan'
      option fallback_relay 'rd dhcpv6 ndp'
      option compat_ula '1'
      option rd 'server'
      option dhcpv6 'server'
      option management_level '1'
      option always_assume_default '1'
      option ndp 'relay'

    root@Metamagix:~# cat /etc/config/network

    config interface 'loopback'
      option ifname 'lo'
      option proto 'static'
      option ipaddr '127.0.0.1'
      option netmask '255.0.0.0'

    config interface 'lan'
      option type 'bridge'
      option proto 'static'
      option netmask '255.255.255.0'
      option ipaddr '192.168.1.1'
      option ifname 'eth0 wdev0ap0 wdev1ap0'
      option ip6assign '64'

    config interface 'wan'
      option ifname 'eth1'
      option _orig_ifname 'eth1'
      option _orig_bridge 'false'
      option proto 'dhcp'

    config globals 'globals'

    config interface 'wan6'
      option proto 'dhcpv6'
      option ifname '@wan'
      option reqaddress 'force'
      option reqprefix '60'

    root@Metamagix:~# ip -6 route

    2001:558:600a:57:5853:c33:9f54:2ddd dev br-lan  proto kernel  metric 256  expires 254951sec
    2601:602:8000:7eb8::/64 dev br-lan  proto kernel  metric 256  expires 254951sec
    unreachable 2601:602:8000:7eb8::/64 dev lo  proto static  metric 2147483647  error -101
    fe80::/64 dev br-lan  proto kernel  metric 256
    fe80::/64 dev eth1  proto kernel  metric 256
    fe80::/64 dev wdev0  proto kernel  metric 256
    fe80::/64 dev wdev1  proto kernel  metric 256
    default via fe80::21d:70ff:feb0:2ee2 dev eth1  proto kernel  metric 1024  expires 1798sec

    root@Metamagix:~# ip -6 neigh

    fe80::21d:70ff:feb0:2ee2 dev eth1 lladdr 00:1d:70:b0:2e:e2 router DELAY

    root@Metamagix:~# ifstatus wan6

    { "up": true,
      "pending": false,
      "available": true,
      "autostart": true,
      "uptime": 89502,
      "l3_device": "eth1",
      "proto": "dhcpv6",
      "device": "eth1",
      "metric": 0,
      "ipv4-address": [ ],
      "ipv6-address": [ { "address": "2001:558:600a:57:5853:c33:9f54:2ddd", "mask": 128, "preferred": 254891, "valid": 254891 } ],
      "ipv6-prefix": [ { "address": "2601:602:8000:7eb8::", "mask": 64, "preferred": 254891, "valid": 254891, "class": "wan6",
                       "assigned": { "lan": { "address": "2601:602:8000:7eb8::", "mask": 64 } } } ],
      "ipv6-prefix-assignment": [ ],
      "route": [ { "target": "::", "mask": 0, "nexthop": "fe80::21d:70ff:feb0:2ee2", "metric": 1024, "valid": 1797 } ],
      "dns-server": [ "2001:558:feed::1", "2001:558:feed::2" ],
      "dns-search": [ ],
      "inactive": { "ipv4-address": [ ], "ipv6-address": [ ], "route": [ ], "dns-server": [ ], "dns-search": [ ] },
      "data": { } }

    root@Metamagix:~# ifstatus lan

    { "up": true,
      "pending": false,
      "available": true,
      "autostart": true,
      "uptime": 89536,
      "l3_device": "br-lan",
      "proto": "static",
      "device": "br-lan",
      "metric": 0,
      "ipv4-address": [ { "address": "192.168.1.1", "mask": 24 } ],
      "ipv6-address": [ ],
      "ipv6-prefix": [ ],
      "ipv6-prefix-assignment": [ { "address": "2601:602:8000:7eb8::", "mask": 64, "preferred": 254861, "valid": 254861 } ],
      "route": [ ],
      "dns-server": [ ],
      "dns-search": [ ],
      "inactive": { "ipv4-address": [ ], "ipv6-address": [ ], "route": [ ], "dns-server": [ ], "dns-search": [ ] },
      "data": { } }

How can I fix this issue?

Thanks in advance!

Chadster766, 2015-09-09:
I've struggled with IPv6 on McWRT. I'm of the opinion that IPv6 implementation shouldn't be done unless your ISP only supports IPv6. My testing shows that the vast majority of the Internet isn't IPv6 ready and that IPv6 doesn't provide any performance enhancements.

That being said IMO the solution to the McWRT IPv6 issue is to upgrade the odhcpd client package and custom configure the firewall for IPv6 ICMP.

Let me know how that goes. Sorry I don't have more time to spend on the IPv6 issue myself at this time.

bikeNomad, 2015-09-10:
I was interested in ipv6 for the sake of remote access via Apple's "back to my Mac", but should be able to get it to work via IPV4. Anyway, thanks for your work on McWRT!.

closed 2015-09-10 by bikeNomad