Chadster766/McWRT #101: opkg can not install ipk

ohyeah521, 2015-09-10:
hi ,i had set up this os, but i can not install program。
for example:
opkg insatll ipset iptables-mod-nat-extra dnsmasq-full

this will show an error.

please help me. thank you.

chocolim, 2015-09-10:
Can you add a screenshot or the complete output please?

bikeNomad, 2015-09-10:
did you do
    opkg update
first?

Also, you must spell "install" correctly (your example did not).

ohyeah521, 2015-09-10:
@chocolim  

root@McWRT:~# opkg update
Downloading http://www.protechs-online.com/downloads/McWRT/packages/Packages.gz.
Updated list of available packages in /var/opkg-lists/mcwrt.

root@McWRT:~# opkg install ipset
Installing ipset (6.11-2) to root...
Downloading http://www.protechs-online.com/downloads/McWRT/packages/ipset_6.11-2_armadaxp.ipk.
Multiple packages (kmod-ipt-core and kmod-ipt-core) providing same name marked HOLD or PREFER. Using latest.
Multiple packages (kmod-ipt-core and kmod-ipt-core) providing same name marked HOLD or PREFER. Using latest.
Configuring luci-app-shadowsocks-spec.
//usr/lib/opkg/info/luci-app-shadowsocks-spec.postinst: line 4: default_postinst: not found
Collected errors:
 * satisfy_dependencies_for: Cannot satisfy the following dependencies for ipset:
 *      kernel (= 3.2.36-1-51098ce98ff1bc265ed1fe337696946b) *  kernel (= 3.2.36-1-51098ce98ff1bc265ed1fe337696946b) *
 * opkg_install_cmd: Cannot install package ipset.
 * pkg_run_script: package "luci-app-shadowsocks-spec" postinst script returned status 127.
 * opkg_configure: luci-app-shadowsocks-spec.postinst returned 127.
root@McWRT:~#

==========================================================

please help me. thank you.

ohyeah521, 2015-09-10:
@bikeNomad   yes, i had did that.

Chadster766, 2015-10-23:
I think I saw an ipset option during the last build. This might be a kernel related option. If try to build the image yourself keep an eye out for the prompt because if I remember correctly it stop there for Y/N. I've been selecting N my releases.

closed 2020-02-14 by Chadster766