Chadster766/McWRT #103: Flash fails.

MasterXX, 2015-10-04:
Whatever I do, I can't get your firmware to work.
I've installed the firmware (Flashed)
Router reboots, and the Power ans eSATA lights come on.
When I try to go to the web interface, (192.168.200.1) I get a time out request.
After that, the router resets it self to the backed up firmware...
I'm lost... 
Flashing stock OpenWRT works.
Please help!

Chadster766, 2015-10-04:
Do you have a version 1 WRT900AC?

JW0914, 2015-10-04:
Have you verified the hash to make sure you're not flashing a corrupted download?  Another user on the OpenWRT WRT1900 thread a few days ago had a myriad of issues and the cause ended up being he had a corrupted download of the firmware.

MasterXX, 2015-10-06:
@Chadster766 I've got the build in fan, so it's the V1. 
@JW0914 How do I check the hash? I see the hash of the copy I downloaded, but where is the original hash to compare it to?

MasterXX, 2015-10-06:
@JW0914 
The hash from the latest file is:
42F28819FB972D5F90F34F7BA804BA33.

MasterXX, 2015-10-06:
Also, this is the only firmware that wouldn't flash.
I've could flash OpenWRT Stock, and Kaloz's build with no problem.
The weird thing is, I have flashed your firmware before.
It worked then. I've reverted to stock, because first I didn't like it.
Now I want to give your firmware a second try, because it is overloaded with features in comparison with the stock OpenWRT firmware.

JW0914, 2015-10-06:
You can install all the components of McWRT on OpenWRT.  This isn't to say don't use McWRT, simply that both can configured with the same packages

MasterXX, 2015-10-06:
@JW0914 But how? It would be a pain in the ass to do that manually.
I also don't know how to prepare a auto install script.

MasterXX, 2015-10-22:
I've got it fixed!! :D I'm a happy McWRT user now.
Solved by flashing the AAv1.0.8 build,
and then flashing the latest build.

closed 2015-10-22 by MasterXX