Chadster766/McWRT #106: 192.168.200.1 No Response/Login

Tez-187, 2015-10-22:
I have had this firmware installed for around 2 weeks everything has been fine but the last few days i now can not login to the router on any device as before i had no issues logging in. How the F do fix this or flash my router back to linksys firmware please help i need access to my router 192.168.200.1 no longer works

Chadster766, 2015-10-22:
The switch from one image to the other can be caused by repeated power cycling.

Tez-187, 2015-10-22:
That may have been it i did turn it on and off 3 times in a row. Im switching back to stock firmware as network transfers and way to slow on your firmware

Tez-187, 2015-10-22:
I just hit restore firmware and it took me back to your firmware so that was great i will continue to use it as it provides me stable WiFi and linksus oem doesn't. How ever network transfer times are a joke do you plan on fixing anytime soon?

Chadster766, 2015-10-22:
That's a big job because many file associated packages need to be updated. I'm not sure when I will have time.

nnorcross, 2015-10-22:
No, worries...

As a Newbie to this area, I mistakenly hit the button...

I have downloaded and installed your work to my router and find it very
nice to work with.

Only question I would have Sir, is where do I control the bandwidth limits
for the wifi?

MasterXX, 2015-10-22:
I have the same error. The router flashes the stock firmware back, after the McWRT firmware has been installed.

closed 2015-10-23 by Chadster766