Chadster766/McWRT #107: Slow network transfer speeds

Tez-187, 2015-10-22:
The network transfer speeds are 10x slower than linksys OEM firmware. Just putting it out there so people know

MasterXX, 2015-10-22:
Chad is already aware of this. It is caused by the driver.

Tez-187, 2015-10-22:
Correct i just put this here so everyone can be aware

Chadster766, 2015-10-23:
In release V2.4 I've updated the USB3 controller driver. Maybe it will make a difference with this issue.

Tez-187, 2015-10-23:
Great i will test when home and let you know

Tez-187, 2015-10-25:
Speeds are still very slow 13 MB/s after version 2.4 update so no gain in performance

Chadster766, 2015-10-25:
Is the drive formatted ntfs and is that the upload or download speed? Is it married over wireless or word?

Tez-187, 2015-10-25:
Yes drive is ntfs and speeds are wired. Transferring from pc to network USB 3.0 drive speeds are 13MB/s and from network drive back to pc is 54 MB/s. Linksys firmware was 70-80MB/s from pc to network drive

Chadster766, 2015-10-25:
Those are the same speeds i see as well.

Tez-187, 2015-10-25:
So is it possible to get the speeds up to linksys standards?

Chadster766, 2015-10-25:
Sure but more work well need to be done. 

Regards,
 
Chad McCuewww.mccuesoft.com
www.protechs-online.com
IT Administrator\Database Administrator\Software and Website Developer
Electrical Class “M” Voice-Video-Data Technician
PLC\Electrical Control Applications Developer
Satellite\Home Theatre TechnicianOffice: 204-364-2096
Cell: 431-283-3021 Sent from my Samsung Galaxy smartphone.

closed 2020-02-14 by Chadster766