Chadster766/McWRT #110: WRT3200ACM corrupt bootloader recovery?

panda-mute, 2017-06-09:
https://github.com/Chadster766/McWRT/wiki/WRT1900AC-corrupt-bootloader-recovery not work in WRT3200ACM.
Is there any method to reflash uboot of WRT3200ACM?

Chadster766, 2017-06-09:
Follow the instruction in the below link in the Armada 385 tabs:

https://wiki.openwrt.org/toh/linksys/wrt_ac_series#corrupt_environment_recovery

panda-mute, 2017-06-12:
I have try to follow the instruction:
./kwboot -t /dev/ttyUSB0 -b u-boot-a38x-spi-uart.bin -d
Sending debug message. Please reboot the target.../
Sending boot image...

Booting from NAND flash

mvBoardIdGet: TWSI Read for Marvell Board ID failed (57) 
	Using default board ID

General initialization - Version: 1.0.0
mvSysEnvIsFlavourReduced: TWSI Read of 'flavor' failed
Detected Device ID 6820
High speed PHY - Version: 2.0

Initialize DB-88F6820-BP board topology

Init DB board default topology
Device 6810/20 supports only 2 SATA interfaces: SATA Port 3 @ lane3 disabled
updateTopologySatR: TWSI Read of 'dbserdes1/2' failed
board SerDes lanes topology details:
 | Lane #  | Speed |  Type       |
 --------------------------------
 |   0    |  03   |  SATA0	|
 |   1    |  05   |  PCIe0	|
 |   2    |  05   |  PCIe1	|
 |   4    |  05   |  USB3 HOST0	|
 |   5    |  05   |  USB3 HOST1	|
 --------------------------------
:** Link is Gen1, check the EP capability 
PCIe, Idx 0: Link upgraded to Gen2 based on client cpabilities 
PCIe, Idx 1: detected no link
High speed PHY - Ended Successfully
DDR3 Training Sequence - Ver TIP-1.32.0
mvSysEnvGetTopologyUpdateInfo: TWSI Read failed
  0 % [+xmodem: Protocol error

It seems the device not into uart boot stage? How to trigger the uart boot?

Chadster766, 2017-06-12:
Try "./kwboot -t /dev/ttyUSB0 -b u-boot-a38x-spi-uart.bin " instead

If you would like to do Skype screen sharing for more detailed troubleshooting you can Skype call me: Skype Name: chad.mccue766

panda-mute, 2017-06-12:
Sorry I don't have Skype now.

Try as your advice:
Sending boot message. Please reboot the target...|
Sending boot image...
  0 % [+++++++++++++++++xmodem: Bad message

Is there any help for this message?

Chadster766, 2017-06-12:
That looks good but it fails in the middle of the uart transfer. 

Check you TTL leads are connecting well.

panda-mute, 2017-06-13:
The TTL connection is well, because it is working with another WRT3200ACM(input and output).
Anyway I change to another TTL to USB cable, the boot is hang, no uboot printout.
So can you advice which TTL to USB cable is work in WRT3200ACM?

Chadster766, 2017-06-13:
I've asked the expert on this process to help with your boot loader recovery:

https://github.com/nitroshift/armada-38x-bootloader-recovery-files/issues/1

sirentatsina, 2017-12-17:
panda-mute

have you solved the "WRT3200ACM corrupt bootloader recovery" yet?
Few days ago, I met the same problem.
Could you please tell me the solution?
Thanks

closed 2020-02-14 by Chadster766