Chadster766/McWRT #111: Bricked WRT1900AC

cgcarlos, 2018-10-21:
Tried the Bootloader recovery as recommended and still fails to boot correctly. I have attached a txt file that shows when the router boots and fails. Is this router done and nothing can help it. When you power on the router you get a constant power led and the sata led is constanly on. Can't access network ports. Let me know if this router is a paper weight.

[WRT1900AC (Bricked).txt](https://github.com/Chadster766/McWRT/files/2498691/WRT1900AC.Bricked.txt)

Chadster766, 2018-10-21:
Try the new instructions and u-boot images in the below wiki:

https://github.com/Chadster766/McDebian/wiki/C.-U%E2%80%90Boot-Recovery

rip057, 2018-10-21:
Your need to get a uart ttl USB cable 4 pin... They cost about 3 dollars
and take apart the router hook up the pins to the cable via the guide and
pray you don't really go wtf at this level cause it gets super hairy....

closed 2020-02-14 by Chadster766